cyphunk/JTAGenum #4: Proper Usage

Brumstar, 2017-11-21:
First of all, thanks for writing this tool.

I just want to verify that I am using this tool properly in my test environment. My current setup is two TI EX-TM4C123GXL boards. One board runs the sketch, and the other board is just for the JTAG interface. The pinout for this board can be found [here](http://energia.nu/wordpress/wp-content/uploads/2014/01/LaunchPads-LM4F-TM4C-%E2%80%94-Pins-Maps-11-32.jpeg). The GPIOs of the first board are connected to the second board, with the proper pin numbers placed in the sketch. 

After successfully connecting to the terminal, I run a loopback check, which gives no results. Then I run a scan, and icodescan. Still nothing. Then I do a boundary scan (mentioned to be broken, can I help fix?) and get some results:

~~~~
> b
================================
Starting shift of pattern through bypass...
Assumes bypass is the default DR on reset.
Hence, no need to check for TMS. Also, currently
not checking for nTRST, which might not work
active  tck:PA5 tdo:PB5 tdi:PE5  bits toggled:12
~~~~

The only one of these that was correct is TDO. Again, I saw that this was mentioned as broken so this is not surprising. However, was I supposed to be getting results for the other commands? I have tried with pullup resistors on and off, as well as the prescale define there or not. I would love to use this project as part of my pentest routine. Can you please tell me what I am doing wrong?

cyphunk, 2017-11-28:
Supposedly (I'm saying this because I've only looked briefly at the [usermanual](http://www.ti.com/lit/ug/spmu296/spmu296.pdf)) you have to "Switch the POWER SELECT switch to the right for Debug mode" according to page 14 of usermanual. Apart from that I'm still trying to understand how this processor and evaluation kit multiplex the JTAG with SWD (aka SWJ) on this "ICDI" port.

Page 20 of usermanual (link above) suggest that for the U1-A processor the JTAG pins are:
* PC0 TCK/SWCLK
* PC1 TMS/SWDIO
* PC2 TDI
* PC3 TDO/SWO

Yet none of those show up in the pin diagram you sent. But further, on page 22 we see that for the processor U2-A it's JTAG pins connect directly to U1-A, in a manner that doesnt appear to permit some sort of pass through. So I'm not sure how JTAG is configured here. 

But you say that JTAGulator finds JTAG, which is interesting. So perhaps I'm really missing some detail

Brumstar, 2017-11-28:
Unfortunately, I can't find a pinout diagram that actually labels the JTAG pinout. If you look at the board pinout picture you can clearly see the JTAG interface. In the [TM4C123GH6PM Microcontroller Data Sheet](http://www.ti.com/lit/ds/spms376e/spms376e.pdf) page 201 you can see the JTAG pinout values as well as additional JTAG information.

TCK - Pin 52 (PC0)
TDI - Pin 50 (PC2)
TDO - Pin 49 (PC3)
TMS - Pin 51 (PC1)

I can attach some screenshots of output from the JTAGulator if you would like that to verify what I said. When I just looked at the board I was using as a target I was in fact using the debug USB power. To give additional information, I had another 3.3V target board that I used with both JTAGulator and this project and JTAGulator was able to find JTAG whereas this was not. However, as this was an unknown target board I decided to verify the JTAGenum code using these known boards. Hope I can help figure this out.

cyphunk, 2017-11-28:
Ah! Thanks for pointing out the labels. Didn't notice them at first. 

I seem to have chased a rabit down a hole: I tried to visually trace the labeled pins to the pins in the datasheet you linked to but they don't line up. Tracing of TMS TDO and TDI can be done visually from the first [pin map image](http://energia.nu/wordpress/wp-content/uploads/2014/01/LaunchPads-LM4F-TM4C-%E2%80%94-Pins-Maps-11-32.jpeg) you linked to. It appears those JTAG lines connect to pins 20 21 and 22. I couldn't find a datasheet for the LX4F120 chip shown in the pin map image but assume the map is the same as for the [TM4C123GH6PM](http://www.ti.com/lit/ds/spms376e/spms376e.pdf) page 1328 indicates pins 20 21 and 22 are PA3 PA4 and PA5. These don't clearly indicate JTAG/SWD/SWJ functionality. And as you pointed out in page 201, and page 1328, JTAG is indicated on pins 49-52.

Okay, I'll assume that maybe the traces in the pin map image really depend on LX4F120, and that for the TM4C123GH6PM the traces will show up visually different and into the correct pins.

thinking...

cyphunk, 2017-11-28:
Okay, a small step backward: Could we consider the voltage of the target and the testing microcontroller?

One of the things JTAGulator does great, if I remember correctly, is handle differing target voltages. Whereas JTAGenum's microcontroller and the Target need to be operating at the same voltage on GPIO's/pins. Could it be that the evaluation kit running JTAGenum is using 5v for GPIO pins while the target is at 3.3v?

Brumstar, 2017-11-29:
In my most basic test setup which I outlined in my first post I was using one EK-TM4C123GXL (oops I put EX in my first post) board to test JTAG on the other. I'm pretty sure they were operating at the same voltage as I did nothing to configure them differently. [This board](http://www.ti.com/tool/EK-TM4C123GXL) seems to have the TM4C123GH6PM chip. Let me know what other information I can provide for you!

cyphunk, 2017-12-03:
There is a slight chance that JTAG IO could operate at a different voltage than digital IO pins, but, probably they are the same. I'm sorry though I can't look into this in detail at the moment :(

Brumstar, 2017-12-19:
Really cool of you to order the boards! I look forward to your test results whenever you get around to it :)

Brumstar, 2018-08-23:
Hey dude. I made a fork https://github.com/Brumstar/JTAGenum. You will see that in the top level there is a file `tivac1.0.3.tar.gz`. I'm pretty sure this is what you're looking for. Let me know!

cyphunk, 2018-08-23:
thanks for that. Yes so now I appear to have some issue with the underlying arm compiler I'm using ``arm-none-eabi-gcc-8.2.0-1``.  Either the downloaded tivac-core or the tar.gz you sent both have this issue ``fatal error: stdint.h: No such file or directory``. I had to modify the platform.txt and edit the flags as mentioned here https://github.com/sandeepmistry/arduino-nRF5/issues/57#issuecomment-260531890

Now its compiling and will if I can load to board

cyphunk, 2018-08-23:
Okay resolve that (had to re-install ``arm-none-eabi-newlib``) then a few more changes to platform.txt and it compiles! yay

cyphunk, 2018-08-23:
So I'm not seeing the same results, no JTAG detected. I will try to use this on another target and will see where I get in the coming days.

Brumstar, 2018-08-23:
Do you mean that jtagenum was actually able to detect JTAG on the interface? Or was it not able to? My original results were that jtagenum could not detect JTAG on an interface that the JTAGulator could.

cyphunk, 2018-08-23:
Sorry, yes I mistyped. I meant that I *am* seeing the same results.

I was able to run the same JTAGEnum code on a Teensy and detect JTAG on a new target I have. But I was unable to detect the JTAG port on the Launchpad master board, from either Launchpad or Teensy. So something is wrong and I'll look at it more as I have time.

Brumstar, 2018-08-23:
Cool thanks for clarifying!

cyphunk, 2018-08-24:
Okay just a small update. I had a pin def wrong and now the JTAGenum on Launchpad finds the same target that the Teensy found. I'm now left wondering what the story is with the Launchpad master boards JTAG. I dont know when I will get to looking at it. Perhaps later, perhaps not. I'm on the clock :/

Here is the output that the Launchpad gave for my target (the target is a IMX6 chip and the ID found is correct):

```
> i
================================
Starting scan for IDCODE...
 ntrst:PE_0 tck:PA_2 tms:PA_4 tdo:PB_2 tdi:PF_0	devices: 1
	0x2191C01D
================================

> r
Pullups ON

> i
================================
Starting scan for IDCODE...
 ntrst:PE_0 tck:PA_2 tms:PA_4 tdo:PB_2 tdi:PF_0	devices: 1
	0x2191C01D
================================

>
> b
================================
Starting shift of pattern through bypass...
Assumes bypass is the default DR on reset.
Hence, no need to check for TMS. Also, currently
not checking for nTRST, which might not work
================================

> s
================================
Starting scan for pattern:0110011101001101101000010111001001
FOUND!  ntrst:PE_0 tck:PA_2 tms:PA_4 tdo:PB_2 tdi:PF_0 IR length: 5
active  ntrst:PE_0 tck:PA_2 tms:PF_0 tdo:PB_2 tdi:PA_4	bits toggled:6
================================

> r
Pullups OFF

> s
================================
Starting scan for pattern:0110011101001101101000010111001001
FOUND!  ntrst:PE_0 tck:PA_2 tms:PA_4 tdo:PB_2 tdi:PF_0 IR length: 5
active  ntrst:PE_0 tck:PA_2 tms:PF_0 tdo:PB_2 tdi:PA_4	bits toggled:6
================================

>
```

cyphunk, 2018-08-24:
Not sure how to get JTAG working on the Tiva C in general. I'm following the instructions [here](https://e2echina.ti.com/cfs-file/__key/telligent-evolution-components-attachments/00-96-00-00-00-11-70-81/Using-TM4C12x-Devices-Over-JTAG-Interface.pdf) on page 13 with no connecting it to OpenOCD. Were you able to connect with JTAG to the target? If so I'd have a couple questions about the proper connectivity

cyphunk, 2018-08-24:
Scratch that, I've been able to connect to the Tiva now with a different JTAG probe (Jlink).

cyphunk, 2018-08-27:
JTAGenum on either a Teensy or Tiva C is able to find another Tiva C. Appears the problem was properly setup to enable JTAG on the Tiva C. After following the instructions on how to enable JTAG on the Tiva C from link before ([page 13 here](https://e2echina.ti.com/cfs-file/__key/telligent-evolution-components-attachments/00-96-00-00-00-11-70-81/Using-TM4C12x-Devices-Over-JTAG-Interface.pdf)) JTAG shows is found. So I suspect what you were seeing with Jtagulator was a false positive that also showed up on JTAGenum as well when the target Tiva C device did not have its JTAG enabled.

Here is the resulting output from one Tiva C scanning another Tiva C:
```
> i
================================
Starting scan for IDCODE...
(assumes IDCODE default DR)
 ntrst:PA_5 tck:PB_1 tms:PE_4 tdo:PE_5 tdi:PB_4  devices: 1
  0x4BA00477
 ntrst:PB_4 tck:PB_1 tms:PE_4 tdo:PE_5 tdi:PA_5  devices: 1
  0x4BA00477
================================

> s
================================
Starting scan for pattern:0110011101001101101000010111001001
active  ntrst:PA_5 tck:PB_1 tms:PB_4 tdo:PE_5 tdi:PE_4  bits toggled:6
FOUND!  ntrst:PA_5 tck:PB_1 tms:PE_4 tdo:PE_5 tdi:PB_4 IR length: 4
active  ntrst:PB_4 tck:PB_1 tms:PA_5 tdo:PE_5 tdi:PE_4  bits toggled:12
active  ntrst:PB_4 tck:PB_1 tms:PE_4 tdo:PE_5 tdi:PA_5  bits toggled:2
================================
```

[photo of the setup](https://imgur.com/a/lnEhIjR), note that I'm using one Tiva as the target and one Tiva as the scanner.

So I'm closing the bug. Happy I could test the Tiva and add it as known supported platform. If you have any further questions, or more issues, feel free to re-open or open new issues. 

For perpetuity sake, I'll discuss why the false positive might appear when the target JTAG is disabled, and why in the above results you see the IDCODE twice.

The IDCODE shows up twice because TRST and TDI are not needed for the IDCODE scan. For the IDCODE scan to show results it assumes that the target has the IDCODE as the default Data Register (DR) between TDI and TDO at reset. The reason the scan is made this way is for two reasons. 1) The JTAG standard itself specifies that either Bypass or IDCODE will be the default DR, and 2) to set the IDCODE as the DR would require knowing the proper length of the Instruction Register (IR). IR length can differ per target so to avoid another brute force scan for IR Length JTAGenum only finds IDCODE when it is already the default DR and assumes user will use the pattern scan as well which would find devices that do not have IDCODE as default.

About the false positive... the pattern scan can result in many false positives when there is a short or crosstalk between wires. This is part of why the loopback test is available. But even a loopback may not always detect target shorts. So one will have to assume some false positives can appear.

closed 2018-08-27 by cyphunk

cyphunk, 2019-01-16:
For reference, using the TI Teva C as an example target requires setting it up to provide JTAG. This is described in [page 13 of this document](https://e2echina.ti.com/cfs-file/__key/telligent-evolution-components-attachments/00-96-00-00-00-11-70-81/Using-TM4C12x-Devices-Over-JTAG-Interface.pdf).

Summary:
* The TevaC has two boards with one that has a header where the JTAG pins (TDO,TMS,etc) are clearly labeled. This will be the target.
* Connect the pin labeled EXT DBG to a GND pin
* Set the Power Select switch to Debug