cyphunk/JTAGenum #12: Teensy 4.0

brandonros, 2019-12-14:
Is Teensy 4.0 supported?

cyphunk, 2019-12-15:
I Teensy 4.0 has support in Teensyduino then it will work without or with very little modification to JTAGenum. I see on the Teensy 4 release that they say Teensyduino can be used to program Teensy 4.0 (_["Teensy 4.0 can be programmed using the Arduino IDE with Teensyduino add-on."](https://www.pjrc.com/store/teensy40.html)_. I  have not had a chance to test it yet.

brandonros, 2019-12-16:
[image]

Can you recommend what pins I would need to connect? I have a 2x5 (10-pin) JTAG header I am trying to reverse engineer with a Teensy 4 and could use a bit of direction if you don't mind.

[image]

cyphunk, 2019-12-22:
This board looks very welcoming to RE. Have you checked if you can find a footprint for the MCU. Because there are so many traces you may be able to map the JTAG pins, or at least map one to confirm it is connected. (which is likely).  Next we would want to check if the MCU has any security features to disable JTAG through fuses, which we may or may not find in the general datasheet.

brandonros, 2019-12-22:
https://www.continental.com/en/press/press-releases/2017-03-28-control-unit-e-class-64576

> Next we would want to check if the MCU has any security features to disable JTAG through fuses, which we may or may not find in the general datasheet.

I believe JTAG is disabled since this is a "production" device but I thought at least IDCODE would work.

I'm not sure if I need some kind of fuse/resistor to enable it. Any thoughts/ideas?

https://www.infineon.com/dgdl/Infineon-TC27xDC-DS-v01_00-EN.pdf?fileId=5546d462694c98b4016953972c57046a

I am under the impression that I can't trace because a) the board is multiple layers and b) the chip is in a special FBGA packaging that hides the pins?

I'd love to pair with you on this. Any advice? I think one mistake I am making is I am putting GND in the pin array, but taking it out yields no difference.

brandonros, 2019-12-22:
[image]

brandonros, 2019-12-22:
[image]

brandonros, 2019-12-22:
I am told there is no external MCU. All of the other chips on the board are auxilliary chips for converting voltage levels from TTL to CAN-H/L etc.

ikarus23, 2020-04-29:
First time JTAGenum user here:
I'm using a Teensy 4.0 at a know JTAG port (dev board). The method IDCODE scan works, but not very reliable. I've noticed that I get better results when downclocking the Teensy to 150MHz instead of 600MHz. Is this normal? Is it more reliable on an Arduino?

OT: 
The output is very verbose for cases where it needn't to be. With IDCODE scan TDI can not be identified. Why printing different pin layouts where only TDI changes?