cyphunk/JTAGenum #14: False positives?

brandonros, 2019-12-16:
```
> pattern scan
================================
Starting scan for pattern:0110011101001101101000010111001001
FOUND!  ntrst:A0 tck:A1 tms:A2 tdo:A8 tdi:A3 IR length: 0
FOUND!  ntrst:A0 tck:A1 tms:A3 tdo:A8 tdi:A2 IR length: 0
FOUND!  ntrst:A0 tck:A1 tms:A3 tdo:A8 tdi:A9 IR length: 0
FOUND!  ntrst:A0 tck:A1 tms:A4 tdo:A8 tdi:A2 IR length: 0
FOUND!  ntrst:A0 tck:A1 tms:A4 tdo:A8 tdi:A3 IR length: 0
FOUND!  ntrst:A0 tck:A1 tms:A4 tdo:A8 tdi:A9 IR length: 0
FOUND!  ntrst:A0 tck:A1 tms:A6 tdo:A8 tdi:A2 IR length: 0
FOUND!  ntrst:A0 tck:A1 tms:A6 tdo:A8 tdi:A3 IR length: 0
FOUND!  ntrst:A0 tck:A1 tms:A6 tdo:A8 tdi:A9 IR length: 0
FOUND!  ntrst:A0 tck:A1 tms:A7 tdo:A8 tdi:A2 IR length: 0
FOUND!  ntrst:A0 tck:A1 tms:A7 tdo:A8 tdi:A3 IR length: 0
FOUND!  ntrst:A0 tck:A1 tms:A7 tdo:A8 tdi:A9 IR length: 0
FOUND!  ntrst:A0 tck:A1 tms:A9 tdo:A8 tdi:A2 IR length: 0
FOUND!  ntrst:A0 tck:A1 tms:A9 tdo:A8 tdi:A3 IR length: 0
active  ntrst:A0 tck:A2 tms:A1 tdo:A3 tdi:A5  bits toggled:75
active  ntrst:A0 tck:A2 tms:A1 tdo:A8 tdi:A5  bits toggled:76
active  ntrst:A0 tck:A2 tms:A1 tdo:A8 tdi:A6  bits toggled:65
active  ntrst:A0 tck:A2 tms:A1 tdo:A8 tdi:A7  bits toggled:77
FOUND!  ntrst:A0 tck:A2 tms:A1 tdo:A8 tdi:A9 IR length: 0
active  ntrst:A0 tck:A2 tms:A1 tdo:A9 tdi:A5  bits toggled:75
FOUND!  ntrst:A0 tck:A2 tms:A3 tdo:A1 tdi:A5 IR length: 0
active  ntrst:A0 tck:A2 tms:A3 tdo:A8 tdi:A1  bits toggled:76
FOUND!  ntrst:A0 tck:A2 tms:A3 tdo:A8 tdi:A9 IR length: 0
active  ntrst:A0 tck:A2 tms:A3 tdo:A9 tdi:A5  bits toggled:2
active  ntrst:A0 tck:A2 tms:A4 tdo:A1 tdi:A5  bits toggled:4
active  ntrst:A0 tck:A2 tms:A4 tdo:A3 tdi:A5  bits toggled:75
FOUND!  ntrst:A0 tck:A2 tms:A4 tdo:A8 tdi:A5 IR length: 0
active  ntrst:A0 tck:A2 tms:A4 tdo:A8 tdi:A6  bits toggled:39
active  ntrst:A0 tck:A2 tms:A4 tdo:A8 tdi:A7  bits toggled:39
FOUND!  ntrst:A0 tck:A2 tms:A4 tdo:A8 tdi:A9 IR length: 0
active  ntrst:A0 tck:A2 tms:A4 tdo:A9 tdi:A5  bits toggled:75
active  ntrst:A0 tck:A2 tms:A5 tdo:A1 tdi:A4  bits toggled:45
active  ntrst:A0 tck:A2 tms:A5 tdo:A1 tdi:A6  bits toggled:41
active  ntrst:A0 tck:A2 tms:A5 tdo:A1 tdi:A7  bits toggled:33
active  ntrst:A0 tck:A2 tms:A5 tdo:A1 tdi:A8  bits toggled:26
active  ntrst:A0 tck:A2 tms:A5 tdo:A3 tdi:A8  bits toggled:75
active  ntrst:A0 tck:A2 tms:A5 tdo:A3 tdi:A9  bits toggled:75
FOUND!  ntrst:A0 tck:A2 tms:A5 tdo:A8 tdi:A3 IR length: 0
active  ntrst:A0 tck:A2 tms:A5 tdo:A8 tdi:A9  bits toggled:4
active  ntrst:A0 tck:A2 tms:A5 tdo:A9 tdi:A3  bits toggled:75
active  ntrst:A0 tck:A2 tms:A5 tdo:A9 tdi:A8  bits toggled:41
active  ntrst:A0 tck:A2 tms:A6 tdo:A3 tdi:A5  bits toggled:75
FOUND!  ntrst:A0 tck:A2 tms:A6 tdo:A8 tdi:A5 IR length: 0
active  ntrst:A0 tck:A2 tms:A6 tdo:A8 tdi:A7  bits toggled:47
FOUND!  ntrst:A0 tck:A2 tms:A6 tdo:A8 tdi:A9 IR length: 34
active  ntrst:A0 tck:A2 tms:A6 tdo:A9 tdi:A5  bits toggled:75
active  ntrst:A0 tck:A2 tms:A7 tdo:A1 tdi:A5  bits toggled:18
active  ntrst:A0 tck:A2 tms:A7 tdo:A3 tdi:A5  bits toggled:75
FOUND!  ntrst:A0 tck:A2 tms:A7 tdo:A8 tdi:A5 IR length: 0
active  ntrst:A0 tck:A2 tms:A7 tdo:A8 tdi:A6  bits toggled:39
FOUND!  ntrst:A0 tck:A2 tms:A7 tdo:A8 tdi:A9 IR length: 0
active  ntrst:A0 tck:A2 tms:A7 tdo:A9 tdi:A5  bits toggled:75
active  ntrst:A0 tck:A2 tms:A8 tdo:A1 tdi:A5  bits toggled:2
FOUND!  ntrst:A0 tck:A2 tms:A9 tdo:A8 tdi:A3 IR length: 0
active  ntrst:A0 tck:A3 tms:A1 tdo:A2 tdi:A5  bits toggled:69
active  ntrst:A0 tck:A3 tms:A1 tdo:A8 tdi:A5  bits toggled:76
FOUND!  ntrst:A0 tck:A3 tms:A1 tdo:A8 tdi:A9 IR length: 0
active  ntrst:A0 tck:A3 tms:A1 tdo:A9 tdi:A5  bits toggled:75
FOUND!  ntrst:A0 tck:A3 tms:A2 tdo:A1 tdi:A5 IR length: 34
active  ntrst:A0 tck:A3 tms:A2 tdo:A8 tdi:A4  bits toggled:20
active  ntrst:A0 tck:A3 tms:A2 tdo:A8 tdi:A6  bits toggled:2
active  ntrst:A0 tck:A3 tms:A2 tdo:A8 tdi:A7  bits toggled:2
FOUND!  ntrst:A0 tck:A3 tms:A2 tdo:A8 tdi:A9 IR length: 0
active  ntrst:A0 tck:A3 tms:A2 tdo:A9 tdi:A5  bits toggled:75
FOUND!  ntrst:A0 tck:A3 tms:A4 tdo:A1 tdi:A5 IR length: 0
active  ntrst:A0 tck:A3 tms:A4 tdo:A2 tdi:A5  bits toggled:75
[... 2081 lines cut ...]
  0x1
 ntrst:A9 tck:A3 tms:A4 tdo:A1 tdi:A8  devices: 1
  0x1
 ntrst:A9 tck:A3 tms:A6 tdo:A1 tdi:A0  devices: 1
  0x1
 ntrst:A9 tck:A3 tms:A6 tdo:A1 tdi:A4  devices: 1
  0x1
 ntrst:A9 tck:A3 tms:A6 tdo:A1 tdi:A7  devices: 1
  0x1
 ntrst:A9 tck:A3 tms:A6 tdo:A1 tdi:A8  devices: 1
  0x1
 ntrst:A9 tck:A3 tms:A7 tdo:A1 tdi:A0  devices: 1
  0x1
 ntrst:A9 tck:A3 tms:A7 tdo:A1 tdi:A4  devices: 1
  0x1
 ntrst:A9 tck:A3 tms:A7 tdo:A1 tdi:A6  devices: 1
  0x1
 ntrst:A9 tck:A3 tms:A7 tdo:A1 tdi:A8  devices: 1
  0x1
 ntrst:A9 tck:A4 tms:A8 tdo:A1 tdi:A0  devices: 1
  0x1
================================
```

It isn't clear to me which pattern is the actual pattern I'm interested in.

cyphunk, 2019-12-19:
I finally had a chance to look at this. 

On the first setion, pattern scan: There are too many FOUND results. Even after I remove the results that could have been due to the loopback entries, there are still too many. It is not uncommon to have many "FOUND" results but typically you would see that at least the TDI, TDO and TCK pins remain constant. Having the correct ntrst pin is not always required and even the TMS pin may not be required if the target is already in bypass mode on reset. On your target in the FOUND results after removing loopback pairs, the TDI and TDO change too often to indicate proper JTAG found. Another issue with the results is that the IR length on the FOUND entries is often ``0``. IR of 0 I think pretty much impossible so those entries can be removed. The remaining entries show very high IR lengths. Most targets have IR length of 3 to 8. The IR length will increase if more chips are in the chain together. So those high IR lengths of 34 and 68 are not impossible but unlikely. For example here are the common irlength's found in targets supported by OpenOCD:

```
openocd-git/tcl$ grep -r "irlen [0-9]" . | sed 's/.*irlen/irlen/'| awk '{print $1" "$2}' | sort | uniq -c | sort -n
1 irlen 11
1 irlen 16
1 irlen 18
1 irlen 2
1 irlen 32
2 irlen 12
2 irlen 7
6 irlen 38
7 irlen 10
18 irlen 8
19 irlen 6
47 irlen 5
197 irlen 4
```

brandonros, 2019-12-19:
I’m working on a Tricore chip. I feel like I read their “miniwiggler” device sets proprietary parameters that enable JTAG on the device.

Here is a rookie question. I have a 10 pin connector but I know one of the pins is GND.  Would I have negative results if I was including that in the possible A1, A2, A3 pin array?

Also, how would I know if I need a resistor before this JTAG connector is usable?

brandonros, 2019-12-22:
```
> loopback check
================================
Starting loopback check...
active  tdo:A1 tdi:A7  bits toggled:38
================================
```

Does this mean I should unplug/unwire with A1, A7, or both?

cyphunk, 2020-01-16:
Sorry for the delay. It depends.  A true loopback would showup as something like ``FOUND!  tdo:A8 tdi:A2 reglen:0``. The loopback tests by running the normal pattern scan where it shifts through a pseudo-random on one pin and checks where it came out. If it finds the pattern it will show up as a ``FOUND!`` entry. This test iterates over pin pairs from all the available pins. When the current under-test pin pair results in bits appear to be toggled by the full pattern was not found it will show up as an ``active`` pair as you see above. This does not mean that it is clearly a loopback. 

You could read your results as "When sending known pattern through pin A1 the state on pin A7 changed between HIGH and LOW 38 times. This could be a result of stimulation sent on A1, could be the result of bad wire insulation on A7, could indicate that A7 is floating between HIGH and LOW on its own or that A7 is pumping out some sort of data or activity independent of stimulation on A1"