cyphunk/JTAGenum #16: No Results

Smokingenius, 2020-04-10:
Greetings,

I am running into an issue using the JTAG enumerator. I originally had found online the rPi version, which linked me back to this page as its source. Unfortunately the rPi version would not enumerate so I had an Arduino and gave that a try, also with no success. 

So, I am practicing my board skills as a hobbyist. I got a pile of boards off eBay, and they all have Spartan 3s on them, the S1000. There is a 6 pin programming header on the board, appears to be JTAG. After getting some advice from the rPi enumerator project, I did trace out vref and gnd on the header, so the remaining 4 are the minimum essential JTAG pins for which are unknown to me. 

I tried the enumerator with the scan command, and nothing arises. I have my wires connected to digital 3-6 on the Arduino. After reviewing the code I saw the notice about having to #define HALFCLOCK for AVR chips, which my Arduino has. I did, same effect, nothing is found. I then tried both versions of the compiled code, going through toggling on the pull up resistor, still nothing is found with scan. I even increased the delay a significant amount, trying each new step, and still no results. 

I am wondering if there is something I am missing. The board has a nicely labeled primary connection header, with board power being via this header. My power supply shows its drawing the correct voltage and amps so I know its on, and its labeled 3.3v, which is what I am supplying. 

Is it possible the binary pattern used to scan is not suitable for FPGA jtag enumeration? 

I did also try with a PIC dev board I have for which I know the JTAG lines, and even then it was not able to find the correct JTAG pin out.

cyphunk, 2020-04-12:
It sounds like you've tried everything I would have tried. I've detected FPGA's before with JTAGEnum. So I'm not sure what could be the issue. If you are able to share some details about the board I would be happy to investigate manuals, other information or even order a similar board in Germany that I could use to try and replicate the issue you found. (feel free to email or post here: nathan _at_  squimp.com )

cyphunk, 2020-04-12:
Would this board from Ebay with a XCS200 be comprable? https://www.ebay.de/itm/XILINX-SPARTAN-3-XC3S200-FPGA-module-FPGA-kit-Development-board-XMF3/182243053522?hash=item2a6e884fd2:g:ZZwAAOSwawpXsH4A

If so I wonder what the significance of the description: " • 2.5V (VCCAUX, JTAG)"

There could be an issue if your Arduino or RaspberryPi are producing signals, and expected results, at 3.3v but the target chip is expected 2.5v. On the other hand it could be 3.3v HIGH and LOW thresholds are perfectly compatible with 2.5v. Here is a voltage level comparison chart for reference:

[image]

(src: https://www.jsykora.info/2014/05/logic-voltage-levels/)

I've not inspected that chart in great detail. It "looks" like 2.5 is close enough to 3.3 but perhaps a closer inspection of this issue would be in order.

let me know about ebay link though and if this is close to your target. If so I'll order and try it out.

Smokingenius, 2020-04-12:
@cyphunk yes that is very comparable chip wise. Its the same chip on two of the 3 practice boards I bought (even the FTG256 sub designation), and the SPI Flash on that board also is in the same family of SPI Flashes that is on my board. 

So for those chips, if you look in the Xilinx pin-out documentation they do mention 2.5v for the VCC Aux lines in for the cells. The SPI flash, and board power, are set to 3.3v, usually, in my experience, and that is what is marked on the boards I have for board power. I was using an external power supply to drive the board power and ground.

I did some extensive troubleshooting this weekend and finally got it to work with the following settings:
-#define for HALFCLOCK on
-delay 100
-pullup on
-target board powered off the Arduino 3.3v and GND lines

That last one is the real key one, as I only started getting correct results when doing this. I eliminated the power supply completely. I did try checking the board draw voltage, and my meter showed 2.95 when running off the Arduino. I set the supply to that but no luck, still didn't work. 

I think the issue is typically with JTAG, in my experience, with FPGAs that vRef value on the JTAG line is important, as that sets the reader on what to expect to and from the JTAG lines. I think by switching to the Arduino powering the target board this provided the missing ref voltage it needed, since it knew exactly how much the Arduino was supplying to the target board. That is my assumption anyways. I tested with another Arduino I found lying around (Its the Engloo Uno? board with the very large, older style Atmel AVR chip on it) and I got the same result and power draw when pulling from board power exclusively. 

I even tried with the supply in the loop, powering the board, with grounds going to the Arduino for ground and JTAG ground, and then the board ground going to the power supply, and that didn't work either. 

Hopefully that helps you all out. Not sure if this is exclusive to FPGAs. I am going to grab some junk boards from work tomorrow that are in the "FREE" bin and see if I get similar results.

Smokingenius, 2020-04-14:
@cyphunk so I have continued testing this week and I can only seem to get results when I am driving the 3.3v line on the dev boards with the 3.3v line from the Arduino power rail. When I drive power with my Rigol programmable power supply, which has digital 30v-30v-5v sections, 5v section and its set to 3.3v, the dev board draws power, but then the Arduino finds a lot of false positives, and of the founds, none are the correct sequence for the pins it discovered previously. Since the dev board has the lines labeled I have been using it for testing and eval purposes. The other two don't but I get the same result regardless. 

I find it very strange the Arduino has to be powering the target to get a proper result. 

This would make it difficult to use if say the target could only be powered by 7v-12v barrel power, or was a unit that only ran off 1.8v.

szymonh, 2020-11-19:
@Smokingenius when powering the target board from the Rigol programmable power supply and trying to enumerate the JTAG pins using Arduino do you have common ground in the circuit? Sorry if the question is silly but my first guess is that an electrical connection between Arduino GND and target GND is missing.