cyphunk/JTAGenum #20: Segmentation fault on raspi model B

elijahlimas123, 2022-04-05:
i have git cloned into https://github.com/cyphunk/JTAGenum.git. i cd into JTAGenum and run sudo bash JTAGenum.sh. this returns with Segmentation fault. prior to getting this error it was telling me that i was missing parentheses in the print statements in the .sh file, i fixed that and now running JTAGenum.sh gives me Segmentation fault.

i am on raspberry pi model B

cyphunk, 2022-04-06:
Could you run the bash script with -x so we can figure out which command is segfaulting.

``bash -x ./JTAGenum.sh``

elijahlimas123, 2022-04-06:
thanks for the response,  i was able to resolve this problem. however, i am now encountering another. the issue is that loopback check, scan and scan_idcode are not returning any results. i have enabled verbose output and i still don't see the problem. any suggestions?

elijahlimas123, 2022-04-06:
my pinout is: 14 22 19 3 23, with pin 14 being ground.

cyphunk, 2022-04-07:
I'm sorry to say that beyond the information that is in the readme and in the wiki I'm not able to get much more advice. Whenever I use JTAGenum I will test it first against a target whose JTAG pins I know for certain are active and responding. This confirms the toolkit is working as I expect. Then I move to testing on an unknown target.

Without a known jtag active device to test against I can recommend checking loopback works by shorting two of the pins. Result should be loop is found there. I can also say that the shell script version has not been as extensively tested. It works in some conditions and might not work in others.