cyphunk/JTAGenum #29: No /sys/class/gpio/gpio${num} with Raspberry PiOS

redawl, 2024-12-03:
It looks like this pin interface has been deprecated in the linux kernel for a while, and has finally been disabled in the latest version of PiOS:
https://www.thegoodpenguin.co.uk/blog/stop-using-sys-class-gpio-its-deprecated/

Any idea how to update JTAGEnum.sh to work with the new interface? 
I took a stab at it, but haven't had any luck so far.

cyphunk, 2024-12-04:
I see 3 potential solutions but would recommend that the first, despite requiring a bit more code change, will be the easiest to complete because it requires the least amount of work on setting up depencies

option 1: install wiringpi. see the [blink.sh example](https://github.com/WiringPi/WiringPi/blob/master/examples/blink.sh). Basically every call to /sys/**/gpio would need to be changed to use the `gpio` binary provided by wiringpi.

option 2: install the fuse library [sysfs-gpio-shim](https://forums.raspberrypi.com/viewtopic.php?t=367548) which is wrapper that provides the /sys/**/gpio interface again. However, it requires libgpiod version 2 which you have to install from sources. libgpiod version 2 is in the [pipeline for rapsberrypi os](https://packages.debian.org/search?keywords=libgpiod) so maybe it can also be installed more easily in future.

option 3: use builtin gpioget, gpioset commands. There's a [discussion here](https://forums.raspberrypi.com/viewtopic.php?t=343514) about some issues around values not being saved which might be worth reading before using this option.

I currently cannot implement or test any of these solutions due to work/life constraints. Would be happy to merge any pull requests though.

redawl, 2024-12-05:
I went with the `sysfs-gpio-shim`, seems to work perfectly, thanks!

closed 2024-12-29 by redawl